ChrisWag91/Inkscape-Lasertools-Plugin #1: lines drawn between sections

jdplus3, 2019-10-03:
Chris, I am having lines engraved as if the laser is not being turned off before travel perhaps 6 lines on a small project connecting words of text. any thoughts? thank you for the software.

pppalain, 2020-01-17:
noticed something similar: in infill, the generated vectors skip many letters but do not request a fast move since it is a vector. i have kept the file with the problem

pppalain, 2020-01-17:
Changing the laser width appears to have made the problem go away. (I know not a fix)

pppalain, 2020-01-17:
saving to png in high res (150 dpi) and then tracing again helps.

ChrisWag91, 2020-03-31:
should be fixed with one of the latest updates. Please validate the fix with your example file.

closed 2020-05-10 by ChrisWag91