ChrisWag91/Inkscape-Lasertools-Plugin #13: Text created in Inkscape doesn't place well.

horga83, 2020-09-08:
Hi Chris,

Awesome job on the plugin, thank you very much.
I have created Text in Inkscape and rendered it with the plugin and while it works, the text often ends up away from the 0,0 creation point, which makes a trial and error when burning.  Always a test burn first.

This is with plugin V1.0 and Inkscape 1.0 running under Linux.

Also in your instructions on the github page you have an extra step 5 under: If you work from svg file.
Should I put enhancement requests in a separate issue or would you like me to contact you directly?

Cheers
George

ChrisWag91, 2020-09-10:
Hi,

0,0 should always be the left lower edge of your page.

[image]

Please send the file for further testing if this is not the case.

Thanks for the heads up regarding the readme file - this is now fixed.

cheers
chris

horga83, 2020-09-10:
Ah Chris it seems to be an issue of whether the object is selected or not.  It may be best to note that the object should NOT be selected in the Readme.

So what I did to be successful was:
Note this is for an SVG file.

1) Load file into Inkscape
2) Resize object to desired size
3) Ungroup until no groups remain
4) Change page size in document properties and make sure units in mm and scale to 1
5) Unselect any objects
6) Render with Lasertools

When I was unsuccessful I skipped step 5 so the object was still selected.
The object still rendered but it was placed way down the page.

ChrisWag91, 2020-09-11:
Thanks for investigating this. 
Sounds like that should be fixable in code.
I'll look into that as soon as I have some time on hand.

ChrisWag91, 2020-09-14:
Hi Horga,

I tried to reprodce the issue with a bunch of different files without any success.
Can you send me your example along with a screenshot of your settings page and/or the log file.

cheers
Chris

horga83, 2020-09-14:
Hi Chris,
This was a case of having the object selected instead of the entire
document.  If  I deselect the text after I stroke to path it works
correctly.Thanks for looking at this.
CheersGeorge

closed 2020-09-17 by ChrisWag91