ChrisWag91/Inkscape-Lasertools-Plugin #14: SVG file doesn't seem to render

horga83, 2020-09-08:
Hi Chris,

I pulled the attached SVG file off the net and I can't seem to get the plugin to render it without first converting to a PNG and then importing into Inkscape.  Thought you might be interested in testing with it.
[Sunflower.svg.zip](https://github.com/ChrisWag91/Inkscape-Lasertools-Plugin/files/5191266/Sunflower.svg.zip)

ChrisWag91, 2020-09-09:
Hi Horga,
Thanks for the feedback.
I had a quick look at the file. It seems that there are some problems with the file itself. For example inkscape does not see any layers in this file.  
Also make sure you have all the units set to mm instead of px and the scale is set to 1.
[image]

But it seems to work just fine if you do that and manually ungroup the object before running lassertools (right click -> ungroup until no more groups are present and then save the file). This will happen automatically if the file is a native inkscape design.
[image]

Can you verify this on your system? 

cheers
Chris

horga83, 2020-09-09:
Hi Chris,
The issue seemed to be keep ungrouping until there are no more groups, I always just hit the ungroup in the toolbar once.  Perhaps you could add that tip to your instructions on the github readme.   Any tips for when placement of the rendered object does not start at 0,0?

Thanks

ChrisWag91, 2020-09-10:
readme is now clarified

closed 2020-09-10 by ChrisWag91