ChrisWag91/Inkscape-Lasertools-Plugin #16: Directory does not exist

Thijsvanzwam, 2020-09-21:
When hitting "apply" in the lasertool tab under extensions, the following error comes up:

Directory does not exist! Please specify existing directory at options tab!

changed the output folder to my desktop but the error remains.
Could I be doing something wrong?

Oh, and I have to use the LaserTools 0.9 version with Inkscape 1.0.1 as the 1.0 version crashes Inkscape to the point of complete reinstall...

I'm running Inkscape on a Mac

dapperfu, 2020-09-21:
I believe you need to have an absolute path.

Thijsvanzwam, 2020-09-21:
path I use is: /usr/username/desktop
see what else I can try...

I use the most recent version, downloaded it today.

dapperfu, 2020-09-21:
I believe OS X home directories are ```/Users/``` not ```/usr/```

https://support.apple.com/en-us/HT201548

Thijsvanzwam, 2020-09-21:
true, checked and it is /Users/ I filled in... typed it wrong in my post

Thijsvanzwam, 2020-09-21:
ok. got it to work. 
used LaserTools.py from the 0.9 version. all other files are from the 1.0 folder.

closed 2021-03-20 by ChrisWag91