ChrisWag91/Inkscape-Lasertools-Plugin #18: LaserTools window too tall

oilytin, 2020-10-27:
Hi the popup is too tall for my screen (1080p) i am unable to see or click the apply button.
Windows 10, Inkscape 0.92
Thanks Anthony
[image]

oilytin, 2020-10-27:
not sure if this tab is the cause of the problem
[image]

oilytin, 2020-10-27:
upgraded to 1.0 and the problem is gone

ChrisWag91, 2020-10-28:
Hi,
I tried to reproduce the issue in windows and linux in v0.92 without any luck.
It displays as intended on my system:

[image]

But the issue seems to pop up in some installations.
It is almost certainly caused by the debug text in your screenshot. 
According to the lasertools.inx file there should not be any text in front of the textboxes:
[image]

Adding a space or a dash in there might fix the problem.
If someone has a chance to test that, please let me know if it solved the problem.

oilytin, 2020-10-29:
Before i moved to the 1.0 version i did try adding something in those quotes but it didnt seem to make any difference with the same output as above.

oilytin, 2020-10-29:
I also tried the 32 and 64 bit .92 releases but that didn't make any difference either.

Lannert, 2020-12-11:
Hi,
i also had the problem with the tall windows (Inkscape 0.92, MacOSX). I added the spaces as recommended and everything is fine. 
Thank you very much for this plugin! It works very good and i enjoy it.

ChrisWag91, 2020-12-12:
Thanks for the Feedback. 
I'll add the spaces in master.

closed 2021-12-12 by ChrisWag91