ChrisWag91/Inkscape-Lasertools-Plugin #19: Laser power variable

flabodotes, 2021-01-27:
Hello @ChrisWag91 ,
  First thanks for your work, I start using it yesterday and I'm getting good results.

That ticket is not an issue but a new feature. With your approach + plugin the G-Code is perfect for  Black and White but what about laser with pulse width modulation (PWM)? My laser could be set in a range of 0-255 witch varies the laser power itself.
Is it possible to analyze the image to apply different grayscale values? Right now only the laser-on-command is executed with always the same intensity, in my case for example to use the 50% power of laser I use M106 S125.

Best Regards

ChrisWag91, 2021-01-28:
Hallo,
That should be possible but it would require quite a bit of additional code.
I might look into that when I get some time on my hand. 
If I need grayscale engraving, I usually generate a halftone pattern. That might not always be a good solution but can look quite good on certain designs.

vinfaraci, 2021-01-28:
There is a really nice raster program for photo engraving called Better Laser Raster for photo engraving. This will have do PWM raster engraving. There is a newer version on the Apple App store http://www.candlercustoms.com/better-laser-raster/