ChrisWag91/Inkscape-Lasertools-Plugin #21: Add Contours generates a closed figure V1.0

Grumpy-Mike, 2021-03-10:
When using just the Add Contours tick box, the Gcode produced seems to turn the objects into a closed figure. This makes spirals not work as they are cut across from start to end. Also lines and curves are converted into closed figures again by joining the start and end points together. So a single line turns into two cuts start to finish and finish to start and a Bessel curve into two closely spaced curves.
Otherwise very well done in getting something to generate Gcode on V1 of Inkscape.

AppsLee, 2021-03-18:
I can second that.
I used a box generator and experienced some issues with this. I found out the cause and fixed it by using only closed figures, but for designs it's a problem.
I used the 0.9 version for Inkscape 0.92.5

ChrisWag91, 2021-03-20:
The plugin expects to work with (closed) paths only. So as of now you would need to convert all objects to paths before you generate the gcode using the "Path/Object to Path" or/and "Path/Stroke to Path" feature.
Unfortunately I have not found a simple way to automate that yet, but I'll look into it when I have some time on hand.