ChrisWag91/Inkscape-Lasertools-Plugin #24: Errors on Execution

631557fcd, 2021-12-09:
Copied files to Extensions folder. Plugin is listed and starts. When clicking on "Anwenden", it seems to do something, but stops with following errors:

`Traceback (most recent call last):
  File "lasertools.py", line 1737, in <module>
    laser_gcode().run()
  File "/snap/inkscape/9256/share/inkscape/extensions/inkex/base.py", line 131, in run
    self.save_raw(self.effect())
  File "lasertools.py", line 1708, in effect
    self.area_fill()
  File "lasertools.py", line 1133, in area_fill
    with Pool() as pool:
  File "/usr/lib/python3.8/multiprocessing/context.py", line 119, in Pool
    return Pool(processes, initializer, initargs, maxtasksperchild,
  File "/usr/lib/python3.8/multiprocessing/pool.py", line 191, in __init__
    self._setup_queues()
  File "/usr/lib/python3.8/multiprocessing/pool.py", line 343, in _setup_queues
    self._inqueue = self._ctx.SimpleQueue()
  File "/usr/lib/python3.8/multiprocessing/context.py", line 113, in SimpleQueue
    return SimpleQueue(ctx=self.get_context())
  File "/usr/lib/python3.8/multiprocessing/queues.py", line 336, in __init__
    self._rlock = ctx.Lock()
  File "/usr/lib/python3.8/multiprocessing/context.py", line 68, in Lock
    return Lock(ctx=self.get_context())
  File "/usr/lib/python3.8/multiprocessing/synchronize.py", line 162, in __init__
    SemLock.__init__(self, SEMAPHORE, 1, 1, ctx=ctx)
  File "/usr/lib/python3.8/multiprocessing/synchronize.py", line 57, in __init__
    sl = self._semlock = _multiprocessing.SemLock(
PermissionError: [Errno 13] Permission denied
`

631557fcd, 2021-12-09:
P.S.: This is whatś in log.txt:

===================================================================
Lasertools log file.
Started at 09.12.2021 17:25:06.
/home/benutzer/proj/Laser/ErsteTests/log.txt
===================================================================

Inserting orientation points 
Applying all transformations 
Flattening beziers 
=================================================================== 
Start generating infill 09.12.2021 17:25:07 
=================================================================== 
Checking directory:  /home/benutzer/proj/Laser/ErsteTests/ 
Testing writing rights on  /home/benutzer/proj/Laser/ErsteTests/output_0003.ngc 
0:00:00.747377     Time until path selection 
 
Working on path:  
0:00:00.072908     Time for csp transformation 
0:00:00.073001     Time for calculating bounds 
0:00:00.000237     Time for calculating zigzag pattern 
0:00:00.000456     Time for rotating 
0:00:08.352621     Time for calculating intersections

ChrisWag91, 2021-12-10:
Can you provide the Inkscape file you used for testing?
You could try to disable multi threading in the preferences since the traceback shows an error in the multi threading code.

631557fcd, 2021-12-10:
Hello and thanks for the fast reply. The error happens on all files. If multi threading is disabled, it works. Which is good so far, but on more complex files it is very slow. I forgot to mention that i use Inkscape V1.1 on Kubuntu 20.04. Thanks for your work by the way!

ChrisWag91, 2021-12-12:
I managed to reproduce this issue. Seems to be caused by missing x permissions on the temp share. 
For a quick test you could try `sudo chmod 777 /dev/shm` in the console.
For a permanent solution check this thread: https://stackoverflow.com/questions/2009278/python-multiprocessing-permission-denied

Please let me know if that solved the issue for you.

631557fcd, 2021-12-13:
Yes, this solves the problem. But as i am a bit paranoid, i think i stick with calculating on a single thread for safety reasons. Thanks again!

closed 2021-12-13 by ChrisWag91