ChrisWag91/Inkscape-Lasertools-Plugin #26: pen plotter

vaclavmuller, 2021-12-20:
Hello

Thank you for this plugin compatible with Inkscape 1.0
I have a question about the usability of the generated g-code for a pen plotter.
I am specifically talking about this type of plotter.

https://www.youtube.com/watch?v=ka2blkdJKEM&ab_channel=MAKERQ
https://www.thingiverse.com/thing:2349232/files

Is there a difference between a pen plotter code and a laser code?

ChrisWag91, 2021-12-20:
Yes, generally that should work just fine. 
You need to change the "Laser ON" and "Laser OFF" parameter to an instruction that actuates the pen.

vaclavmuller, 2022-01-05:
Could I switch the Pen Up and Pen Down commands?

I need this:
Pen (Servo) Up Command: M03 S80
Pen (Servo) Down Command: M05 S0

ChrisWag91, 2022-01-05:
You should be able to just put your gcode in these fields:
[image]

vaclavmuller, 2022-01-05:
If I am right, there I can define the amount of motor rotation.
But how to change the "M" parameters?
The normal command for Pen Up is M05, for Pen Down it is M03,
but my plotter needs these "M" parameters to be switched. 
M03 for Pen UP, M05 for Pen Down.

https://www.youtube.com/watch?v=I4omT2L9aI8&t=542s