ChrisWag91/Inkscape-Lasertools-Plugin #29: Big Y-offset problem

BCChad, 2022-02-09:
My path is set anchored in the lower-left at 0,0 x,y coordinates on the Inkscape page, and the page is set to the same size of the object.
On compiling the GCode, the X coordinate is correct but the Y is off by -246mm.
The computed strokes generated by your Extension appear in that position, -246mm below the bottom of the Inkscape page.

It does give a warning about omitting a non-path, which I cannot find.  I selected everything and converted objects to path.

ChrisWag91, 2022-02-12:
Hi Chad,
can you provide the inkscape file in question for testing?

cheers

BCChad, 2022-02-12:
Yes, here's the file, both SVG and GCODE.   I just ran it again and it again warned about omitting a non-path element.
On the first attempt I got an error that didn't repeat on the second attempt.   Do you have a debug setting so I can send you all of the parameters I used for generating the gcode
[image]
[WoodYou_0010.txt](https://github.com/ChrisWag91/Inkscape-Lasertools-Plugin/files/8054726/WoodYou_0010.txt)
?

ChrisWag91, 2022-02-13:
Thanks. I tested your file with a current version of Inkscape and Lasertools and it seems to work fine.
Please check if you are using Inkscape 1.x with Lasertools for Inkscape 1.x from this repo.
There could also be a problem with your inkscape configuration.
My page is set up like this:
[image]

Here is the svg File after generating the gcode:
[image]

BCChad, 2022-02-13:
Thank you.  Yes I am using version 1.1 of Inkscape.  I notice there now is a 1.1.1 and 1.1.2 available - Should I upgrade to that?

I have confirmed the settings, and also the instructions you posted on the github page.
I'm still getting a "Warning:omitting non-path" consistently - not sure what else is lurking in my .SVG file as I have ungrouped, selected all, and converted objects to paths.

I did discover this - if the objects are all selected, the OFFSET problem occurs (shifted down y -246).  If I de-select the objects, the overlay occurs correctly on top of the text.  Perhaps this finding is helpful to you.

Your instructions do say dependencies are python-lxml, python-numpy - I could not find them anywhere in the inkscape installation;  where do they install?

JuPrgn, 2023-12-02:
I add a problem similar and scale in page settings was not 1.0, changing scale to 1.0 fixed the huge Y offset

osgooda2-ai, 2025-10-20:
Hello all 
I have an issue after I convert the image that when I cut the image it tends to be larger than my set sizes. Meaning that if I have a 6x6 square and a 6x6 image inside that after the gcode is generated it actually cut outside the 6 inches. 
I have also checked the calibration of my cnc on all axis. They are fine. Any ideas? #39