ChrisWag91/Inkscape-Lasertools-Plugin #36: I get an error when executing the script

khvalera, 2023-08-06:
This extension works with Paths and Dynamic Offsets and groups of them only! All other objects will be ignored!
Solution 1: press Path->Object to path or Shift+Ctrl+C.
Solution 2: Path->Dynamic offset or Ctrl+J.
Solution 3: export all contours to PostScript level 2 (File->Save As->.ps) and File->Import this file.

This extension works with Paths and Dynamic Offsets and groups of them only! All other objects will be ignored!
Solution 1: press Path->Object to path or Shift+Ctrl+C.
Solution 2: Path->Dynamic offset or Ctrl+J.
Solution 3: export all contours to PostScript level 2 (File->Save As->.ps) and File->Import this file.

This extension works with Paths and Dynamic Offsets and groups of them only! All other objects will be ignored!
Solution 1: press Path->Object to path or Shift+Ctrl+C.
Solution 2: Path->Dynamic offset or Ctrl+J.
Solution 3: export all contours to PostScript level 2 (File->Save As->.ps) and File->Import this file.

This extension works with Paths and Dynamic Offsets and groups of them only! All other objects will be ignored!
Solution 1: press Path->Object to path or Shift+Ctrl+C.
Solution 2: Path->Dynamic offset or Ctrl+J.
Solution 3: export all contours to PostScript level 2 (File->Save As->.ps) and File->Import this file.

/home/user/.config/inkscape/extensions/Lasertools/lasertools.py:1557: DeprecationWarning: inkex.deprecated.main.transform_mul -> Use @ operator instead
  transf = Transform(transf) * Transform(node.get("transform", None))
This extension requires at least one selected path.

This extension works with Paths and Dynamic Offsets and groups of them only! All other objects will be ignored!
Solution 1: press Path->Object to path or Shift+Ctrl+C.
Solution 2: Path->Dynamic offset or Ctrl+J.
Solution 3: export all contours to PostScript level 2 (File->Save As->.ps) and File->Import this file.

Please select at least one path to engrave and run again.

Traceback (most recent call last):
  File "/home/user/.config/inkscape/extensions/Lasertools/lasertools.py", line 1744, in <module>
    laser_gcode().run()
  File "/usr/share/inkscape/extensions/inkex/base.py", line 250, in run
    self.save_raw(self.effect())
                  ^^^^^^^^^^^^^
  File "/home/user/.config/inkscape/extensions/Lasertools/lasertools.py", line 1737, in effect
    self.bb_preview()
  File "/home/user/.config/inkscape/extensions/Lasertools/lasertools.py", line 1442, in bb_preview
    self.export_gcode(preview_gcode, filename)
  File "/home/user/.config/inkscape/extensions/Lasertools/lasertools.py", line 451, in export_gcode
    f.write(self.header + "\n" + gcode + self.footer)
            ^^^^^^^^^^^
AttributeError: 'laser_gcode' object has no attribute 'header'
sys:1: ResourceWarning: unclosed file <_io.TextIOWrapper name='/home/user/Gcodeoutput_bounding_box.ngc' mode='w' encoding='UTF-8'>

khvalera, 2023-08-06:
It would also be good to make your application multilingual, I can translate it into Ukrainian

vaclavmuller, 2023-08-08:
I can translate it into Czech.

khvalera, 2023-08-09:
An example of how to make a multilingual plugin can be found here: https://github.com/giacmir/Inkscape-JPEG-export-extension

simond98, 2023-09-02:
I kind of solved this problem by changing the character "*" with "@" in the lasertool.py file, line 1557. still it doesn't work, but I don't know why at this point. it says
Can not write to specified file!
/home/simone/Scrivania/prova/prova.ngc